doppelgrau/esp32-f9p-io-board #28: Rotate PCB Labels

flywire, 2019-09-17:
The following labels are inverted relative to others:
- ESP-F9P-IO-Board
- Version 2.2
- GND / 12V / M1A / M1B
- Uart / RX / TX

doppelgrau, 2019-09-17:
Done on purpose.
Assumes mounting with the antenna on top might be likely => easier to read

flywire, 2019-09-17:
A linked built photo would demonstrate.

doppelgrau, 2019-09-17:
Currently work in progress :)

doppelgrau, 2019-10-14:
[image]
Foto from the previous version before rotating the writing. looks wrong when mounted with the cables pointing down (less dust enters case)

closed 2019-10-14 by doppelgrau

flywire, 2019-10-14:
Clearly demonstrated. An easy rotate all labels command would have been nice.